Xinyuan-LilyGO/T-HMI #1: unable to program t-hmi device

GastonLagaffe2013, 2023-02-19:
I have selected Arduino IDE 2.x with "ESP32S3 Dev Module", 16MB flash, hughe app. I use a simple sketch opening the serial port and writing to it. Compilatoin is fine, upload ok but the serial moniro only shows this:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201e6fa
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8

Sketch:
`void setup() {
  Serial.begin(115200);
  Serial.println("Hello T-HMI");
}

void loop() {
  Serial.print(".");
  delay(100);
}
`
this is wierd

lbuque, 2023-02-20:
@GastonLagaffe2013 You try the following configuration, does it work:

[image]

GastonLagaffe2013, 2023-02-20:
Thanks, that fixes the issue

closed 2023-02-20 by GastonLagaffe2013