Xinyuan-LilyGO/T-HMI #3: Grafics problem

369Martin369, 2023-03-09:
try now differnet examples...but failed with any graphics.....eg.; LV:Benchmark.ino result in error:
-----------------------------------
In file included from c:\users\marti\documents\arduino\libraries\lvgl\src/misc/lv_log.h:16,
                 from c:\users\marti\documents\arduino\libraries\lvgl\lvgl.h:25,
                 from c:\Users\Marti\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
                 from C:\tmp\T-HMI-master\T-HMI-master\examples\lv_benchmark\benchmark\lv_demo_benchmark.h:16,
                 from C:\tmp\T-HMI-master\T-HMI-master\examples\lv_benchmark\lv_benchmark.ino:11:
c:\users\marti\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
                  ^~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
----------------------

I installed all 4 libraries!!! does not work.

lewisxhe, 2023-03-10:
Do you use Arduino IDE? You need to add lv_ Conf. h can only be used normally after being copied to the lvgl directory
https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/examples/lv_benchmark/lv_conf.h

lewisxhe, 2023-03-10:
If you need to use TFT_ For eSPI, you need to reference the header file of HMI

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup207_LilyGo_T_HMI.h

369Martin369, 2023-03-11:
> Do you use Arduino IDE? You need to add ` lv_conf. h` can only be used normally after being copied to the lvgl directory https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/examples/lv_benchmark/lv_conf.h

yes, the file is in the folder...as it is your branch

369Martin369, 2023-03-11:
yes I use Arduino IDE.
nExt error is:
----------------------------------------------------
C:\tmp\T-HMI-master\T-HMI-master\examples\lv_benchmark\lv_benchmark.ino:192:21: error: 'esp_lcd_new_panel_io_i80' was not declared in this scope
     ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));
                     ^~~~~~~~~~~~~~~~~~~~~~~~
-----------------------------------------------------

so what libraries are these?
*#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_rgb.h"
#include "esp_lcd_panel_vendor.h"**

thanks for further help.

lewisxhe, 2023-03-12:
This is the internal display driver library of Espressif, not external. Please update the Arduino ESP32 core to the latest

lewisxhe, 2023-03-12:
I have updated all the sample structures, and now it is suitable for ArduinoIDE. Please pull it to the warehouse again and operate according to the steps in the homepage README

369Martin369, 2023-03-14:
can you please create a User_Setup_Select.h file which works with all 3 (T-Embed, T-Display-S3, T_HMI). 
It's annoying to swap libs! Thanks

369Martin369, 2023-03-14:
just confirm that you got my last idea. then I close. thanks

lewisxhe, 2023-03-15:
Please give me some time. I am busy recently. I will add it later

closed 2023-03-16 by 369Martin369