Xinyuan-LilyGO/T-HMI #6: Micropython

Valadimirovich, 2023-04-02:
Will there be a version of Micropython for this board with drivers from lhttps://github.com/russhughes ?

lewisxhe, 2023-04-21:
Sorry, there are currently no suitable ones.
But there is a similar one, hope it can help you
https://github.com/russhughes/st7789s3_mpy

closed 2023-04-21 by lewisxhe

lesept777, 2023-04-28:
From RussHughes:

For the display, try the S3LCD_OCT_16M firmware from https://github.com/russhughes/s3lcd. 

Instructions for flashing:
Place device unto flash mode by pressing and holding the BOOT button, then pressing and releasing the RESET button and finally releasing the BOOT button.

Erase flash
esptool.py --chip esp32s3 --port /dev/ttyACM0 erase_flash

Flash firmware
esptool.py --chip esp32s3 --port /dev/ttyACM0 write_flash -z 0 firmware.bin

Power Cycle device

For the touch screen, I use: https://github.com/fantasticdonkey/uFT6336U. But currently, it's not working (I2C library, SPI touch device)

lesept777, 2023-05-12:
I actually wrote a driver for the touch device, see here:
https://github.com/lesept777/xpt2046_micropython