Xinyuan-LilyGO/T-HMI #7: Test Offset when rotation is set to 3.

TiltedMag830, 2023-05-05:
The text is offset from the side of the screen when setRotation is set to 3, the background fills the full screen. I tried tft.setAddrWindow to update the window size.
~~~
#include "pins.h"

#include <TFT_eSPI.h>
#include <OneButton.h>
#include <Arduino.h>

TFT_eSPI tft = TFT_eSPI();
OneButton buttonOK(21, false, false);

void setup() {
    pinMode(PWR_ON_PIN, OUTPUT);
    digitalWrite(PWR_ON_PIN, HIGH);

    Serial.begin(115200);
    delay(1000);

    
    buttonOK.attachClick(power_off);
    buttonOK.attachLongPressStop(power_off);
    // pinMode(21, INPUT)
    Serial.println("setup");
    
    tft.begin();
    tft.setRotation(1);
    tft.setSwapBytes(true);
    tft.setTextSize(1);
    tft.fillScreen(TFT_GREEN);
    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    
    pinMode(PWR_EN_PIN, OUTPUT);
    digitalWrite(PWR_EN_PIN, HIGH);
    pinMode(BK_LIGHT_PIN, OUTPUT);
    digitalWrite(BK_LIGHT_PIN, HIGH);

    tft.drawString("Booting", 0, 0, 2);
    delay(3000);
}

void power_off() {
    digitalWrite(PWR_ON_PIN, LOW);
    Serial.println("power_off");
    digitalWrite(BK_LIGHT_PIN, LOW);
}

void loop() {
    // buttonOK.tick();
    // delay(10);
    // digitalWrite(BK_LIGHT_PIN, HIGH);
    // tft.setRotation(1);
    tft.setAddrWindow(20,0,240,320);
    for(int i = 0; i<5; i++) {
        tft.setRotation(i);
        Serial.println(i);
        tft.fillScreen(TFT_GREEN);
        tft.setTextColor(TFT_MAGENTA);
        tft.drawNumber(millis(), 0, 100, 4);
        delay(2000);
    }
}
~~~

fozi, 2024-06-25:
I'm having the same issue, it looks like it's in the initialization code, but it's a bit above my level to debug it. Any chance this got solved.

Thanks.

fozi, 2024-06-25:
If anyone needs a fix for this issue, make the following change.

In file: "Setup207_LilyGo_T_HMI.h"
Located in: \libraries\TFT_eSPI\User_Setups

Either remove or remark line 11 "#define CGRAM_OFFSET"

After making that change all rotations orientations work.

teastainGit, 2024-06-25:
Oh, yeah it's always the CGRAM_OFFSET.
Just kidding, amazing find/fix.
I saw you over on the community site, btw.