Xinyuan-LilyGO/T-HMI #8: No touch AND screen on T-HMI. However no problem using SPI ILI9341. How come?

StrandedAlien, 2023-05-25:
Real pity, such a nice piece of HW and half useless. Also had a problem to run the original TFT_eSPI examples at all and found a solution...set PIN 10 to high. Does not help with the touch, see my issue which i already added to  TFT_eSPI  in Bodmers repository.  I do not know whether it is the board or the code. So please have a look at my description of the error:

https://github.com/Bodmer/TFT_eSPI/issues/2669

Separately used, programs with touch work and prog's with screen work, but not together.

lewisxhe, 2023-05-26:
1. You first need to follow the prompts in the [README](https://github.com/Xinyuan-LilyGO/T-HMI/blob/d398bf1bdfee7530cc811f82cc1c65598f669ca5/README.md?plain=1#L27)

2. Refer to the provided example, because it has different places, such as IO10 https://github.com/Xinyuan-LilyGO/T-HMI/blob/d398bf1bdfee7530cc811f82cc1c65598f669ca5/examples/tft/tft.ino#L11

3. If your screen is not displayed, you should check whether HMI is selected here https://github.com/Xinyuan-LilyGO/T-HMI/blob/d398bf1bdfee7530cc811f82cc1c65598f669ca5/lib/TFT_eSPI/User_Setup_Select.h#L134

4. If all the above are normal, then finally you need to refresh the [firmware](https://github.com/Xinyuan-LilyGO/T-HMI/tree/master/firmware) we provide to test whether the hardware is normal

StrandedAlien, 2023-05-26:
Thanks for the fast answer. Did everything as described except Number 4. Because i am on Linux i can't use this flash program. It is windows only. Any suggestions for a workaround?
 

Gesendet: Freitag, 26. Mai 2023 um 04:07 Uhr

lewisxhe, 2023-05-27:
Use esptool ,see [there](https://github.com/Xinyuan-LilyGO/LilyGo-Document/blob/master/MacFlashBin.md)

StrandedAlien, 2023-05-27:
Thanks a lot. The T-HMI specific examples from the root folder all work now. The TFT_eSPI examples using a screen need PIN 10 set to high or blackscreen. In addition Bodmers examples using a touchscreen(TS) , need to be heavily modified in order to use the T-HMI special touchscreen lib to get TS to work. As the delivered T-HMI TS examples are only about calibration and raw data i would like to ask you Lewis for an application TS example. Showing how to use the TS xpt2046.h lib to read some screen presses on rectangles (with round corners) or circles. Also, i have no idea how to use the special TS lib xpt2046.h in LVGL. Is that possible at all? Cause without TS LVGL is a two legged horse.

lewisxhe, 2023-05-30:
xpt2046 is only responsible for providing the pressed coordinates, it cannot recognize a circle or a rectangle, it must be used with lvgl

teastainGit, 2023-05-30:
I admit that I must not understand the Issue here!
Compared to the LilyGO LilyGO T-Display S3, the T-HMI is really well designed and supported. Mine worked first time out of the box and I wrote a bunch of test sketches in Arduino IDE, v 2.1 and using Touch and  TFT_eSPI graphics. Sketches with ESP_NOW, temperature sensor and initially just a simple Touch and graphics demo.
If you want to see the sketch, let me know.
-Terry

StrandedAlien, 2023-05-30:
Hello Terry. 

Probably not even an issue, but my bad understanding. I also asked Bodmer and he said, that i must use a separate library for touch with the T-HMI board. Plain TFT_eSPI simply does not because PIN10 is not set high and it expects the touch on the same SPI line. That is  the reason for using a separate touch library. Wthout it, as i said, it does not work.

And yes, if you have some more sketches. which i could learn from, i would love to get the download links. Thanks for the offer.

teastainGit, 2023-05-30:
[T_HMI_HotHead_TFTandTouch.zip](https://github.com/Xinyuan-LilyGO/T-HMI/files/11600501/T_HMI_HotHead_TFTandTouch.zip)
I am using the current TFT_eSPI from Bodmer's GitHub, version=2.5.30, with the User_Setup_Select.h set for #207 and commented out the line "//#include <User_Setup.h>  
  // Default setup is root library folder"
The Sketch includes #include "pins.h", which references PIN 10 for the #define PWR_EN_PIN  (10).

lbuque, 2023-05-30:
What problems have you encountered? Can't tft and touch be used at the same time?

teastainGit, 2023-05-30:
@lbuque I have encountered no problems with this board and Touch and TFT_eSPI work very well together.

StrandedAlien, 2023-05-31:
I figured it out now. Using an unmodified version of Bodmers TFT_eSPI does not work. The Lilygo display is connected parallel and expects Pin 10 to be set high. You also need a separate lib for touchscreen, else...