Xinyuan-LilyGO/T-HMI #9: Micro sd

telemac73, 2023-07-16:
Hello,

You provide only 3 different pin for tf card but need mimimum 4 ?

Thanks

lewisxhe, 2023-07-16:
1 bit mode, so only 3 IOs are required. Please see the example

telemac73, 2023-07-17:
I try to have it workin in micropython and no example

telemac73, 2023-07-18:
I have also test with the example and i get same error i test with 4 different sd card:

E (2173) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (2173) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107
E (2174) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  2275][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
SD init failed
➸ No detected SdCard

cyril1116, 2023-11-11:
Hi @telemac73,
have you figured it out?

cyril1116, 2023-11-11:
This is needed:
https://github.com/orgs/micropython/discussions/10598

telemac73, 2023-11-14:
Thanks i will try )

cyril1116, 2023-11-14:
@telemac73 
I am already trying it... I built my own firmware (based on clean micropython 1.21.0 and esp-idf 5.0.2) with changes that @dotnfc suggested in https://github.com/orgs/micropython/discussions/10598 but sd card is still not working... I hope I will figure it out and then I will inform you :)

aokhrimenko, 2023-11-22:
Guys, you probably have to power on the peripheral from which SD card is powering as well. 
PWR_EN (GPIO 10): output, high

cyril1116, 2023-11-22:
@aokhrimenko thank you!!! You are absolute legend and I am totally stupid :D You can see in that other discussion I mentioned above, that I tried everything, I even flashed back original firmware from here and tried Arduino testSD sketch and I am probably totally blind because I completely missed fact that in that code, they trigger gpio10 to 1... I even checked schematic many times and yea I am blind... -_-   Ok, so now it works even with micropython - slightly modified micropython - https://github.com/orgs/micropython/discussions/10598#discussioncomment-7545637

vednildip, 2024-11-08:
This display  has  good  at  SD  card  functionality .  TFT_eSPI  library  also  worked  fine  to  draw & Fetch  JPG  image  from  SD  card . But   touch  has  certain  issue  to  display  next  scroll. It  gives  ghost  touches  continuously  of  same  value  and  not  being  integrated  with  TFT_eSPI  library. Please  help  to  resolve  touch  library