Xinyuan-LilyGO/T-HMI #10: STEP 3D model

bojanpotocnik, 2023-07-19:
Hello @LilyGO,

It would be extremely helpful to have a STEP file for easier integration of the T-HMI board in the device design. Would it be possible for you to upload the STEP file of the T-HMI board?

Thank you for considering this request.

lewisxhe, 2023-07-27:
There are currently no models available.

nasaspb, 2023-08-23:
[Lilygo_T-HMI.zip](https://github.com/Xinyuan-LilyGO/T-HMI/files/12421700/Lilygo_T-HMI.zip)
I made the model in Blender. Maybe this will help you.

StrandedAlien, 2023-10-30:
Cool, thanks
[image]

[Lilygo_T-HMI2.zip](https://github.com/Xinyuan-LilyGO/T-HMI/files/13199689/Lilygo_T-HMI2.zip)