Xinyuan-LilyGO/T-HMI #12: image jpg bmp

taoloo897, 2023-08-17:
Hi,
Witch tools did you use to make image .h from jpg?
I found something (Img2Lcd) but not transform ok.
Thanks

nasaspb, 2023-08-23:
Hi.
I used (https://lvgl.io/tools/imageconverter).
Settings:  CF_RGB565A8,  C array, Dither images.
The image must be loaded vertically (portrait orientation).
Maybe it's not right, but I did it.

teastainGit, 2023-08-23:
// ImageConverter 565 Online
//http://www.rinkydinkelectronics.com/t_imageconverter565.php
// On my LilyGO T-Display S3 
//I recommend (170X170 horiz/vert)  or as needed for proportion
//select output as .c file and change just the file extension  to .h---> very important
// Generated from : hothead.jpg
// Time generated : Sat, 01 Oct 22 00:53:37 +0200  (Server timezone: CET)
// Image Size     : h155 x v170 pixels  (vertical max170!!!) fits my LilyGO T-Display S3 
// Memory usage   : 52700 bytes

then in my sketch:
#include "hothead.h"
then in my setup or loop:
tft.pushImage(165, 0, 155, 170, hothead);
-Hope this helps, Terry