Xinyuan-LilyGO/T-HMI #14: Library doesn't work with arduino IDE

jifop, 2023-10-10:
Fails with error "invalid library no headers files"

teastainGit, 2023-10-10:
Can you post your sketch and I will try it on mine?

jifop, 2023-10-11:
no sketch required, if I add the T-HMI library folder to my computer arduino IDE doesn't recognise it as a library.

teastainGit, 2023-10-11:
did you copy all the files and folders from the T-HMI LilyGO GitHub repository 'Lib'  folder into your libraries folder, and overwrite (replace) the existing files?

jifop, 2023-10-12:
I did yes am getting the error below.

Invalid library found in C:\xxx\Arduino\libraries\T-HMI-master: no headers files (.h) found in C:\xxx\Arduino\libraries\T-HMI-master

I've got the board working but would prefer it to be properly installed, currntly cannot work out how to get the bluetooth working, spec says its BT 5.0 but I can't seem to get BT classic to work

StrandedAlien, 2023-10-29:
Try this
Remove T-HMI folder from libraries, instead directly extract it into ARDUINO folder. (Backup and) delete your present TFT_eSPI folder and replace it with the TFT_eSPI folder from T-HMI. Then remove the NOOB trap by setting PIN10 to high (else black screen!) as follows

Replace in every TFT_eSPI example void setup(void) with

#define STONER 10
void setup(void)
{
  pinMode(STONER, OUTPUT);
  digitalWrite(STONER, HIGH);

and no black screen anymore. Making the touch screen examples of TFT_eSPI work is more difficult.