Xinyuan-LilyGO/T-HMI #15: factory example not work on lilygo t-display-s3

taoloo897, 2023-10-23:
I bought this item from aliexpress and it arrive with and example loaded on it and it works ok with next button witch changes the pages.

https://www.aliexpress.com/item/1005005719816754.html?spm=a2g0o.order_list.order_list_main.49.322f1802oPFKyW

After other loaded schetches i loaded from t-hmi library the factory example. it remain in first page with FC CE image shows, the button not works and it remain in this status on serial monitor:

19:39:02.970 -> Turn on the main power
19:39:02.970 -> Power on peripherals, such as the LCD backlight
19:39:02.970 -> Initialize the power button
19:39:02.970 -> Initialize Intel 8080 bus
19:39:02.970 -> Install LCD driver of st7789
19:39:02.970 -> Initialize LCD driver of st7789
19:39:03.102 -> Turn on the LCD backlight
19:39:06.081 -> Initialize touch screen calibration data
19:39:06.081 -> point_1:
19:39:06.081 ->   raw_x: 411  raw_y: 1246
19:39:06.081 -> point_2:
19:39:06.081 ->   raw_x: 284  raw_y: 207
19:39:06.081 -> point_3:
19:39:06.081 ->   raw_x: 320  raw_y: 223
19:39:06.081 -> point_4:
19:39:06.081 ->   raw_x: 342  raw_y: 224
19:39:06.081 -> Initialize the touch screen
19:39:06.081 -> Initialize lvgl

Please help me with your adive what it is wrong in sketch. Thanks in advance.

teastainGit, 2023-10-23:
To test the huge Factory code I had to copy all of the contents of the T-HMI 'lib' directory and copy to my 'libraries' folder and overwrite the existing.  
The device shows FCC, then asks to calibrate the four points of the touch screen.
I press reset and it re-boots as this.
[image]
Hope this is helpful!
-Terry

taoloo897, 2023-10-24:
very strange , i selected other board type "esp32-S3-box" and it works, before i left what arduino ide detected board "lilygo t-dislpay-s3"

teastainGit, 2023-10-24:
Oh, the Arduino IDE has a bug that detects a different board, based on registered types, not necessarily the correct board for you! Annoying and confusing.
All LilyGO S3 boards use:
"ESP32S3 Dev Module" as the board type.
Glad you got it working!
https://github.com/Xinyuan-LilyGO/T-HMI  👍 

[image]

taoloo897, 2023-10-24:
Thanks. Good luck!

closed 2023-10-24 by taoloo897