Xinyuan-LilyGO/T-HMI #18: Lot of errors compiling tft example

andreavalle, 2023-11-11:
Hi, I'm testing the T-HMI with Arduino IDE 2.2.1 and get a lot of errors when compiling the tft example from T-HMI-master.
I have copied all the folders and file from "lib" to my Arduino libraries folder.
`In file included from /xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:20:
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:96:8: error: 'gpio_get_level' was not declared in this scope
   96 |   b  = gpio_get_level((gpio_num_t)TFT_D0); // Read three times to allow for bus access time
      |        ^~~~~~~~~~~~~~
In file included from /xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:96,
                 from /xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:16:
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushBlock(uint16_t, uint32_t)':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:543:5: note: in expansion of macro 'tft_Write_16'
  543 |     tft_Write_16(color);
      |     ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:554:23: note: in expansion of macro 'tft_Write_16'
  554 |   else while (len--) {tft_Write_16(color);}
      |                       ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushSwapBytePixels(const void*, uint32_t)':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:564:20: note: in expansion of macro 'tft_Write_16'
  564 |   while ( len-- ) {tft_Write_16(*data); data++;}
      |                    ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushPixels(const void*, uint32_t)':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:574:37: note: in expansion of macro 'tft_Write_16'
  574 |   if(_swapBytes) { while ( len-- ) {tft_Write_16(*data); data++; } }
      |                                     ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:436:32: note: in expansion of macro 'GPIO_CLR_REG'
  436 |       #define tft_Write_16S(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 0)); WR_H; \
      |                                ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:575:27: note: in expansion of macro 'tft_Write_16S'
  575 |   else { while ( len-- ) {tft_Write_16S(*data); data++;} }
      |                           ^~~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:225:22: error: 'GPIO' was not declared in this scope
  225 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:80:5: note: in expansion of macro 'CS_L'
   80 |     CS_L;
      |     ^~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::begin_nin_write()':
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:225:22: error: 'GPIO' was not declared in this scope
  225 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:92:5: note: in expansion of macro 'CS_L'
   92 |     CS_L;
[... 181 lines cut ...]
      |                  ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:5372:30: note: in expansion of macro 'tft_Write_16'
 5372 |               while (tnp--) {tft_Write_16(textcolor);}
      |                              ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:356:26: error: 'GPIO' was not declared in this scope
  356 |     #define GPIO_CLR_REG GPIO.out1_w1tc.val
      |                          ^~~~
/xxx/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:432:31: note: in expansion of macro 'GPIO_CLR_REG'
  432 |       #define tft_Write_16(C) GPIO_CLR_REG = GPIO_OUT_CLR_MASK; GPIO_SET_REG = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~~~~~~~~~
/xxx/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:5374:19: note: in expansion of macro 'tft_Write_16'
 5374 |             else {tft_Write_16(textcolor);}
      |                   ^~~~~~~~~~~~

exit status 1
Compilation error: exit status 1`

Thanks in advance if someone has an idea of what's going on...
Andrea

lewisxhe, 2023-11-18:
Reduce the esp32 core version to below 3.0.0

andreavalle, 2023-11-19:
Cool thanks! Downgraded to ESP32 2.0.14 and it compiles without errors.

closed 2023-11-19 by andreavalle