Xinyuan-LilyGO/T-HMI #21: ESPHome

c3p0vsr2d2, 2023-12-12:
How can I get this working in EspHome/Homeassistant?

bektorkhan, 2024-04-03:
You can't, not without incorporating TFTeSPI in ESPHome. T-HMI uses 8-bit data to ST7789V and some strange pin assignments . I have like you been struggling, but I'll look into using TFTeSPI arduino lib with ESPHome.

bektorkhan, 2024-04-04:
I got it to work with help of this: https://github.com/landonr/lilygo-tdisplays3-esphome/tree/main
You need to run ESPHome by commandline, Warning: Bat_ADC is used as Reset and GPIO43 as RD did not work otherwise. Both Reset and RD are strapped high in Hardware. And I thought this could be a hardware mod.. but it worked without mod of the board. Modifying this file: 
 .esphome\external_components\6d17677d\components\tdisplays3\display.py :
CONFIG_SCHEMA: This should be set in display def in ESPHome but...
CONF_HEIGHT, default=320
CONF_WIDTH, default=240
Changing the buildflags:
    cg.add_build_flag("-DTFT_INVERSION_OFF")
    cg.add_build_flag("-DTFT_RD=43")
    cg.add_build_flag("-DTFT_D0=48")
    cg.add_build_flag("-DTFT_D1=47")
    cg.add_build_flag("-DTFT_D2=39")
    cg.add_build_flag("-DTFT_D3=40")
    cg.add_build_flag("-DTFT_D4=41")
    cg.add_build_flag("-DTFT_D5=42")
    cg.add_build_flag("-DTFT_D6=45")
    cg.add_build_flag("-DTFT_D7=46")
EspHome:
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
    board_build.f_flash: 80000000L
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  project:
    name: esphome.web
    version: '1.0'

external_components:
  - source: github://landonr/lilygo-tdisplays3-esphome
    components: [tdisplays3]

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  variant: esp32s3
  framework:
    type: arduino

display:
  - platform: tdisplays3
    id: my_display
    height: 320
    width: 240
    update_interval: 1s
    rotation: 90
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(yourcolor));
      it.rectangle(0, 20, it.get_width(), it.get_height(), id(yourcolor));   // header bar

vednildip, 2024-11-12:
ESP32-S3 not supported in IRRemote library? https://github.com/Arduino-IRremote/Arduino-IRremote
in search:- the ESP32-S3 appears to have Remote Control Transceiver (RMT) -ESP documentation that perhaps the ESP32-S3 natively has support for receiving IR signals. is it  ?  or  any suggest  any  other  library

c3p0vsr2d2, 2025-04-05:
> I got it to work with help of this: https://github.com/landonr/lilygo-tdisplays3-esphome/tree/main You need to run ESPHome by commandline, Warning: Bat_ADC is used as Reset and GPIO43 as RD did not [...]

this does not work for me - do you have an updated yaml by any chance that works ?
I can get WiFi and Bluetooth working, but not the display.