Xinyuan-LilyGO/T-HMI #22: Installing T-HMI

RFoulds, 2023-12-20:
I am trying to follow the installation process described in the Readme file.

It says search for the ESP-ARDUINO SDK and install it.

This differs from the T-Display S3 that says to install the ESP32S3 DEV MODULE.

IS ESP-ARDUINO SDK a generic description for one of the options that appear on the menu?

teastainGit, 2023-12-20:
[RFoulds](https://github.com/RFoulds) I agree this makes no sense.
It uses Arduino ESP32 boards  and the final selection is ESP32S3 Dev Module, just like LilyGO T-Display S3 !
I am going to posy my own Issue her to ask LilyGO for clarification(!!!).
-Terry

RFoulds, 2023-12-20:
Thank you for the clarification.  I will happily use the ESP DEV MODULE.

ONE MORE QUESTION.  I have the T-Display s3 working with ESP dev module,
but just noticed that I am now offer the choice of Lilygo T-Display module.
Is that new?

When I copied the lib files for the T-HMI to the Arduino libraries
directory  I did not write over TFT_eSPI or any files that were there for
the T-Display.

The T-HMI does not run any of the examples or any of my T-Display code.
Should I copy and replace the files?

I am ordering 40 displays this month to use with robots for kids with
muscular dystrophy.

Thanks for your help.

Rick

RFoulds, 2023-12-20:
By the way, these are terrific products.

teastainGit, 2023-12-20:
RFoulds:
When the LilyGO T-Display S3 came out in summer 2022 the support was abysmal, I was forced to figure it out and I made my own GitHub repository at:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
RVolos found that the ESP32S3 Dev Module board type would work and so we ran with it! 
Be certain that you are specing  " ESP32S3 Dev Module  "  exactly and it works perfectly. 
The new custom board def LilyGO T-Display S3 is now a little redundant! Too little, too late!

The TFT_eSPI from the Arduino IDE library CAN be used...IF...you keep a copy of the 'User_Setup_Select.h' for each board and copy and replace the existing one in TFT_eSPI library each time you compile for different boards!!!
It can be hand edited by commenting-uncommenting, but it is tedious.

-Terry

RFoulds, 2023-12-20:
Terry,

Thank you very much.

So the trick is that each board
's TFT_eSPI is different with the difference being the User_Setup_Select.h.

That makes perfect sense now.

I am impressed with Volos' videos. They are really great.

Thanks again for your help.

I will run upstairs and try setting up the T-TMI.

I am eagerly waiting for my T-Display Pro to arrive.  That looks perfect
for another part of my exoskeletons.

Rick

RFoulds, 2023-12-21:
Terry,

Could I bother you one more time?
I copied the entire lib contents for the T-HMI to libraries.
I compiled the tft.ino example and it ran.  Progress.

But, I get a message
Invalid library found in
C:\Users\foulds\Documents\Arduino\libraries\arduino_xpt2046_library:
Missing 'paragraph' from library

This does not seem to affect tft.ino, but none of my sketches will run.
Could this message be the cause?

Thanks again,

Rick

teastainGit, 2023-12-21:
Hey Rick:
The xpt2046 library is the touch screen driver and it all works in my environment, right now:

Arduino IDE 2.2.1
Selected as ESP32S3 Dev Module
I deleted the xpt2046 library and  copied it from the LilyGO download, which was just now re-freshed.
I have a test sketch if you are interested, it is slightly diagnostic, in that it demos all the features. It also was written to demo TFT_eSPI and Touch.
Let me know if you want it!
-Terry

RFoulds, 2023-12-21:
Terry

I would greatly appreciate the sketch.  I will follow your instructions.

As I mentioned before, I am doing volunteer work on exoskeletons for young
men with muscular dystrophy.

By age 15, their muscles are so weak that they can't use their arms. Their
hands still work.

I developed the prototype while I was on the faculty at New Jersey
Institute of Technology.  After retirement, I volunteered to help move the
system to a small company in Michigan.  They built 27 pairs of exoskeletons
and set them up with users.  Only 3 pair ever worked.  The, company then
collapses.

I could not abandon the guys and reengaged to fix all the problems.  I am a
one-man operation, doing all software development, mechanical design and 3D
printing, installation and packing.

One major problem is the company's cabling. They are all hand made and are
prone to failure.  I can replace 4 homemade cables with a T-Display S3
Touch and one commercially available USB  cable.

The motor control needs to be tuned for each user.  The company developed a
phone app that fails most of the time with no feedback to warn of the
failure.  I will use the T-HMI or the Pro for easy tuning.

I really appreciate all of your help.  When I write up the project, I will
be sure to acknowledge your assistance

Happy Holidays,

Rick

By the way, where are you located?

RFoulds, 2023-12-21:
Terry,

I am still using IDE 1.8xx since I was somewhere that Lilygo boards didn't
like 2.xx.

I will load the same version as you are using.

Here is a link to my exoskeleton work.

https://drive.google.com/file/d/1pXmmsqOCYC6YFbx5GteHd3_w8wRRWUmM/view?usp=drivesdk

Zach is 27 years old and has not been able to lift his arms for about 12
years.  You can see that  we restore most of his lost range of motion.

Rick

teastainGit, 2023-12-22:
Hi Rick! I'm in Cambridge ON, CA
I am a retired Industrial Robotics Interface Designer, I designed the entire Robotic Interface for Nachi Robots installed at Chrysler in Brampton, to build the LX98 Intrepid cars!
So, I am very interested in exoskeletons.
I could not access the google drive without the password!
Here is my T-HMI demo sketch.
[T_HMI_HotHead.zip](https://github.com/Xinyuan-LilyGO/T-HMI/files/13747415/T_HMI_HotHead.zip)

Nice to talk to you!
-Terry

RFoulds, 2023-12-22:
I will resend the video tonight.
Thanks for the example.  We have lots in common and I hope to pick your
brain about robots.

Rick

RFoulds, 2023-12-22:
Terry,
Sorry for the problem with the video.  The permissions somehow were reset.
This should work.

https://drive.google.com/file/d/1gyZQVYIYGHG2VCyToBo6XYUvv6k2ovy3/view?usp=sharing

These are endpoint exoskeletons. The mechanism is a pair of commercially
available video camera supports.  Each arm has two parallelograms with a
gravity counterbalancing spring.  You can see one alongside the user, with
the other behind the wheelchair.

I added a 10 N-m servo motor to each parallelogram.  The user applies
whatever force he has available (usually 5% of normal) to the cuff.  This
force is sensed by a load cell.  I use an admittance control algorithm to
convert the user force to the desired velocity of the wrist.  This
essentially allows an effective gain of 20-50.

Once set up properly, the user regains nearly full range of motion and can
lift 1 Kg with each hand.

I have recalled all 27 units and am in the process of upgrading them.  Some
upgrades are safety-related.  The 3D-printed brackets fractured on about
half of the systems.  There was no safety switch. The company cut corners.

Some upgrades are performance-related.  I am using the Lilygo products to
provide a display that indicates the modes of operation that are selected.
I added cooling fans since the single-board computer overheated in the
summer.  I also had to develop a way to pull the exo arms in toward the
user so the wheelchair could pass through residential doorways.

Rick

RFoulds, 2023-12-22:
I may not get to try your example until after Christmas as things are
becoming hectic.

RFoulds, 2023-12-22:
By the way, I am in Pennsylvania, US, just north of Philadelphia.

teastainGit, 2023-12-22:
Very cool rig! 
Lets go off line to Email, so as not to clog up the Issue!
We blew through Penn coming back from D.C !!
Here is my email:
teastain@me.com
I will delete most of my comments on this issue, to de-clutter!
-Terry