Xinyuan-LilyGO/T-HMI #27: screen goes black and usb serial becomes unstable after enabling 1-bit microsd

cosmicpsyop, 2024-04-26:
when enabling 1-bit microsd in a custom app with the microsd inserted the display goes black and the serial port becomes unstable and doesn't allow flashing firmware after.

the t-hmi can be recovered by entering the upload mode manually before uploading firmware: i.e.  press and hold the BOOT button and press RST button once, release the RST then release the BOOT button

i returned to the examples and after formatting 3 different fat32 microsd cards, an 8, 16, and 32 gb  but none could be recognized using the factory app and then used examples/sd/sd.ino example and reproduced the failure.

aware of and read [this issue](https://github.com/Xinyuan-LilyGO/T-HMI/issues/9)

lewisxhe, 2024-05-23:
Post your test code? I test

cosmicpsyop, 2024-05-23:
thanks for your reply.  i am just using the examples/sd/sd.ino from this repo to test.

lewisxhe, 2024-05-25:
sd.ino My test here is normal. The SD card will not affect the upload of sketches. You can try rewriting the binary file to see if it runs normally.
https://github.com/Xinyuan-LilyGO/T-HMI/tree/master/firmware

cosmicpsyop, 2024-05-26:
Thanks for looking into this.  

I have a solution to the T-HMI not detecting the sdcard by reducing the SD_MMC frequency.

First, a few comments:
- not sure why the system became unstable when testing sdcard from the custom app, but not an issue with sd.ino.
- not sure why a different result in your environment. perhaps, a different esp_idf version (v5.0.2)?
- the same behavior was observed when using the example factory.ino

So, here is the output for the call *before* the SD_MMC the fix: 

```C
bool rlst = SD_MMC.begin("/sdcard", true);
```

- without sdcard inserted the return code is 0x107

```
E (6044) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (6045) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  6149][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
SD init failed
➸ No detected SdCard
SD close
```

- with the sdcard inserted  the return code is 0x109 and fails

```
E (26223) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (26223) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
[ 26328][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x109). Make sure SD card lines have pull-up resistors in place.
SD init failed
➸ No detected SdCard
```

After poking around SD_MMC.h I tried different frequencies.  Both frequency 5000 and 20000 work while the default of 40000 does not work.  Here is the output of the call working:

```C
bool rlst = SD_MMC.begin("/sdcard", true, false, 20000);
```

- without the sdcard inserted the return code is 0x107
```
(427144) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (427145) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[427250][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
SD init failed
➸ No detected SdCard
SD close
```

- with the sdcard inserted

```
SD close
SD init success
➸ Detected SdCard insert: 7.40 GB
SD close
```

Here is the original code with the fix and a small adjustment to better see the log messages from the console:
```C
#include "pins.h"
#include <Arduino.h>
#include <SD_MMC.h>
#include <FS.h>
#include <SPI.h>
#include <SD.h>

void setup() {
    pinMode(PWR_ON_PIN, OUTPUT);
    digitalWrite(PWR_ON_PIN, HIGH);
    Serial.begin(115200);
    delay(1000);

    pinMode(PWR_EN_PIN, OUTPUT);
    digitalWrite(PWR_EN_PIN, HIGH);
    SD_MMC.setPins(SD_SCLK_PIN, SD_MOSI_PIN, SD_MISO_PIN);
}

void loop() {
    delay(5000);
    bool rlst = SD_MMC.begin("/sdcard", true, false, 20000);
    if (!rlst) {
        Serial.println("SD init failed");
        Serial.println("➸ No detected SdCard");
    } else {
        Serial.println("SD init success");
        Serial.printf("➸ Detected SdCard insert: %.2f GB\r\n", SD_MMC.cardSize() / 1024.0 / 1024.0 / 1024.0);        
    }
    delay(15000);
    Serial.println("SD close");
    SD_MMC.end();
}

```