Xinyuan-LilyGO/T-HMI #28: Constant ghost touches with touch example but not with test firmware

Dakkaron, 2024-09-02:
When I run the touch and touch_calibration examples (as is, no modifications) the tests show constant random touches.

When I upload the test firmware, all works fine.

What could be the problem?

lewisxhe, 2024-09-04:
Erase all flash, re-run the example, and recalibrate

Dakkaron, 2024-09-04:
I did, but that didn't help.

I kinda worked around it by modifying the library.

The ghost touches always have a Z of 0 or 65535 (probably -1, but the variable is uint16_t). Real touches always have other values. So if I filter that out, I don't have ghost touches.

vednildip, 2024-11-08:
screen(serial and tft) continuously in loop showing touch values random even if not touched . s this hardware issue or code. please . how you filtered

vednildip, 2024-11-14:
Hi Dakkaron,
 How  did you  filtered  ghost touches, as   my  T  HMI  having  same  problem ,  as  you  rightly  said  upload the test firmware, all works fine.  but  after  running  the touch and touch calibration examples (as is, no modifications  except  set Brightness(16)) the tests show constant random touches. X  raw  , Y Raw  , X  , Y  and  non  stop  on  serial  monitor.
You  have  also  mention  "if I filter that out, I don't have ghost touches."    Could  you  please  share  that  function  to  filter   X  raw  , Y Raw  , X  , Y  and  non  stop   please  .

lewisxhe, 2024-11-14:
I found that the IRQ was not initialized as input. You can try again. I did not get the ghosting.

Dakkaron, 2024-11-14:
The xpt2046.cpp supplied by this repo here already has the filtering capability in there, but it's commented out.

So I uncommented it and replaced the constant Z_THRESHOLD with a user-definable variable.

[xpt2046.zip](https://github.com/user-attachments/files/17745083/xpt2046.zip)

To enable filtering, you need to call `void setZThreshold(int16_t threshold);` once before checking for touches.

The higher the threshold, the less sensitive the touchscreen will become. Too high of a threshold and you won't register inputs, too low and you'll get more ghost touches.

But @lewisxhe's solution makes sense as well.

lewisxhe, 2024-11-15:
> thanks I will work on press time , serial print result single press "23:01:58.859 -> Pressed! 23:01:58.859 -> RAW X: 1481 RAW Y: 1413 23:01:58.859 -> X: 0 Y: 243 23:01:58.859 -> RAW X: 1481 RAW Y: 1413 23:01:58.859 -> Pressed! 23:01:58.859 -> RAW X: 1487 RAW Y: 1413 23:01:58.859 -> X: 0 Y: 243 23:01:58.859 -> RAW X: 1487 RAW Y: 1413 23:01:58.859 -> Pressed! 23:01:58.859 -> RAW X: 1484 RAW Y: 1413 23:01:58.859 -> X: 0 Y: 243 23:01:58.859 -> RAW X: 1484 RAW Y: 1413 23:01:58.892 -> Pressed! "

Was it resolved after a break fix, or was it fixed using the patch provided by @Dakkaron ?

vednildip, 2024-11-15:
Hi  Lewisxhe **It  was resolved  after "your IRQ" a break fix.**    I  have  just  started learning  C & C++   still premature to  give  my  opinion  but  large  scope  of  improvement in  your library.  I  am sorry  I  am  very  small  to  comment  on  your  work   but  take  it  as  positive  feed back.

Then  I also  tried  other  method " Dakkaron  "  but  had  error  in  compilation saying  XPT2046 is class  can't declare  as  void  for  below  function   also  compilar  says    _zThreshold  is not  within  scope  or  declared  
void XPT2046::setZThreshold(int16_t threshold) {
  _zThreshold = threshold;
}  and  I  was  failed  to  call  function  as  well  setZThreshold(int16_t threshold).

lewisxhe, 2024-11-15:
No, I didn't write the library. I'm just the maintainer.

vednildip, 2024-11-15:
Ohh,    but  you  are  doing  great  service  to  humanity  I know   you  are  expert  in  this field   and  god  has  given  you  such  talented  brain  that's  enough for  student s  like  us . Some  goods  quote  for  you :-  
" Survival is not  aim  , aim  is transformative  success"     What  change  you  bring  to  this word  is more  important  than  win or  loss or  earn  or  gain . 
Keep  it  up  !