Xinyuan-LilyGO/T-HMI #29: Cant flash

Zynth9999, 2024-10-14:
I keep getting this error in esptool:
`A fatal error occurred: Unable to verify flash chip connection (No serial data received.).`

lewisxhe, 2024-10-14:
Try the following steps
[image]

Zynth9999, 2024-10-14:
Will try and post an update

Zynth9999, 2024-10-14:
```
"C:\Users\Janek\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6/esptool.exe" --chip esp32s3 --port "COM20" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x0 "C:\Users\Janek\AppData\Local\Temp\arduino\sketches\D91E7C3EAC94FB9D10797C98D684CEFC/BareMinimum.ino.bootloader.bin" 0x8000 "C:\Users\Janek\AppData\Local\Temp\arduino\sketches\D91E7C3EAC94FB9D10797C98D684CEFC/BareMinimum.ino.partitions.bin" 0xe000 "C:\Users\Janek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\Janek\AppData\Local\Temp\arduino\sketches\D91E7C3EAC94FB9D10797C98D684CEFC/BareMinimum.ino.bin" 
esptool.py v4.6
Serial port COM20
Connecting...Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1064, in _main
  File "esptool\__init__.py", line 859, in main
  File "esptool\cmds.py", line 466, in write_flash
  File "esptool\util.py", line 37, in flash_size_bytes
TypeError: argument of type 'NoneType' is not iterable

Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:a9:dd:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
[8284] Failed to execute script 'esptool' due to unhandled exception!
Failed uploading: uploading error: exit status 1
```
Looks like a problem with setting the flash size

lewisxhe, 2024-10-14:
Try the espressif upload tool and upload the firmware we provide. Try it
https://github.com/Xinyuan-LilyGO/T-HMI/tree/master/firmware

Zynth9999, 2024-10-14:
```
D:\flash_download_tool_3.9.5_0\flash_download_tool_3.9.5
D:\firmware.bin
1 kill
test offset :  0 0x0
case ok
test offset :  0 0x0
case ok

Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
NO XMC flash  detected!
SPI_BOOT_CRYPT_CNT 0
SECURE_BOOT_EN False
Compressed 1229040 bytes to 688625...
[2024-10-14 11:37:07,920][ESP8266Loader_spi[1]][espDownloader.py][line:774][ERROR]: ESP32S3 Chip flash download error esp_write_flash.

```
[image]

Still get errors

lewisxhe, 2024-10-14:
Did you connect any sensors?

Zynth9999, 2024-10-14:
Nope, just the barebones board

lewisxhe, 2024-10-14:
If no sensor is connected, it may be damaged. Contact your dealer.

Zynth9999, 2024-10-14:
Ok will do

Zynth9999, 2024-10-14:
The seller i bough it from doesn't see my order, but i can send you a detailed picture of the board

lewisxhe, 2024-10-14:
What pictures? I can't handle after-sales issues, I can only handle technical issues.

From the above conversation, it can be judged that there is a problem with the board, you need to contact the dealer for replacement

Zynth9999, 2024-10-14:
No i mean is there something wrong with the chip? A missing resistor or something

Zynth9999, 2024-10-14:
Ah ok well sorry for wasting your time with this

closed 2024-10-14 by Zynth9999