Xinyuan-LilyGO/T-HMI #30: Hardware interface(s) Examples  &  Pin  config  Missing

vednildip, 2024-11-04:
Hardware interface(s) Examples  Missing     hence  for  New  starter  it  is difficult  
I2C
I2S
SPI
Native USB
UART
PWM
ADC
GPIO
USB CDC
Also  LovyanGFX  versatile for  MJPG/ JPG  codac    interfacing  pins can  not  identifying  
Also  Audio   A2DP  examples   Missing

vednildip, 2024-11-06:
ArduinoGFX /  LovyanGFX  need  write (RS) and  read  pins (RD ) , reset  pin  using  below  Liligo_T_HMI schematic  show  
CS 6
RS 7
WR 8 
IRQ 9
RESET  -1  (  Not  Defined ) 
But  display  does  yield result , is some thing  wrong  in  my  understanding  while  using   ArduinoGFX /  LovyanGFX

lewisxhe, 2024-11-07:
These examples are basically provided by the official website, but this board has few available GPIOs
[image]

vednildip, 2024-11-07:
Would some  one  help  to  identify  write  and read pins (RD ?) for  T-HMI   Or  simply  share  pins config  for  ArduinoGFX / LovyanGFX  please

vednildip, 2024-11-07:
LovyanGFX   help  me  to  fill  appropriate pins for LilyGo  T-HMI    

class LGFX : public lgfx::LGFX_Device {

  lgfx::Panel_ST7789 _panel_instance;
  lgfx::Bus_Parallel8 _bus_instance;

  public:

    LGFX(void)
    {
      {
        auto cfg = _bus_instance.config();
        cfg.freq_write = 30000000; // used 16MHz and 20MHz during testing
        cfg.pin_wr =  ??DC_PIN; ?????????????????????????????????
        cfg.pin_rd =??  Not_defined ;??????????????????????????????
        cfg.pin_rs = ?? PCLK_PIN ;?????????????????????????

        cfg.pin_d0 = LCD_DATA0_PIN;  // D0
        cfg.pin_d1 = LCD_DATA1_PIN;  // D1
        cfg.pin_d2 = LCD_DATA2_PIN;  // D2
        cfg.pin_d3 = LCD_DATA3_PIN;  // D3
        cfg.pin_d4 = LCD_DATA4_PIN;  // D4
        cfg.pin_d5 = LCD_DATA5_PIN; // D5
        cfg.pin_d6 = LCD_DATA6_PIN;  // D6
        cfg.pin_d7 = LCD_DATA7_PIN; // D7
        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();
        cfg.pin_cs           =    CS_PIN ;
        cfg.pin_rst          =   RST_PIN ;
        cfg.pin_busy         =    -1;

        cfg.panel_width      =   240;
        cfg.panel_height     =   320;

        cfg.readable         = true;
        cfg.invert           = false;
        cfg.rgb_order        = false;
        cfg.dlen_16bit       = false;
        cfg.bus_shared       = true; // SD = true
        _panel_instance.config(cfg);
      }
      setPanel(&_panel_instance);
    }
   auto cfg = _light_instance.config();

      cfg.pin_bl = BK_LIGHT_PIN;
      cfg.invert = false;
      cfg.freq   = 44100;
      cfg.pwm_channel = ??; ???????????????????????????????????

      _light_instance.config(cfg);
      _panel_instance.light(&_light_instance);
    }
    setPanel(&_panel_instance);
  }
};

________________________________________________________________________________________________________________________
++++++++++++++++++++++++++++++++TFT_eSPI  PINS  LilyGO T HMI ++++Refernce +++++++++++++++

#pragma once

#define PWR_EN_PIN  (10)
#define PWR_ON_PIN  (14)
#define BAT_ADC_PIN (5)
#define BUTTON1_PIN (0)
#define BUTTON2_PIN (21)

// lcd
#define LCD_DATA0_PIN (48)
#define LCD_DATA1_PIN (47)
#define LCD_DATA2_PIN (39)
#define LCD_DATA3_PIN (40)
#define LCD_DATA4_PIN (41)
#define LCD_DATA5_PIN (42)
#define LCD_DATA6_PIN (45)
#define LCD_DATA7_PIN (46)
#define PCLK_PIN      (8)
#define CS_PIN        (6)
#define DC_PIN        (7)
#define RST_PIN       (-1)
#define BK_LIGHT_PIN  (38)

// touch screen
#define TOUCHSCREEN_SCLK_PIN (1)
#define TOUCHSCREEN_MISO_PIN (4)
#define TOUCHSCREEN_MOSI_PIN (3)
#define TOUCHSCREEN_CS_PIN   (2)
#define TOUCHSCREEN_IRQ_PIN  (9)

// sd card
#define SD_MISO_PIN (13)
#define SD_MOSI_PIN (11)
#define SD_SCLK_PIN (12)

#define SDIO_DATA0_PIN (13)
#define SDIO_CMD_PIN   (11)
#define SDIO_SCLK_PIN  (12)

vednildip, 2024-11-13:
would you  at least  recommend  IR receiver  compatible  library  to  T HMI  so  I  can  omit  touch  functionality . AS  all  available    IR receiver  library are  not  compatible  with  ESP32 S3  or  not  of  universal  remote.

lewisxhe, 2024-11-13:
This is the IRRemote library compatible with esp32s3
https://github.com/tonhuisman/IRremoteESP8266/archive/refs/heads/feature/C++20-compatibility.zip

vednildip, 2024-11-13:
No my dear IRremoteEsp8266 is not compatible S3. Nor Ir not your touch
library working
 With lot of hope I started my this good display but it is not good. Only
usable with good display

lewisxhe, 2024-11-14:
If the touch screen does not work properly, it is very likely that there is something wrong with the screen. The IRRemote library I sent is compatible with esp32s3 because I am also using this library

vednildip, 2024-11-14:
Thanks for your reply, I found following IR working with simple library by
schreibfaul1/Esp32-IR-Remote-Control

Also I had again burn firmware in your library folder. Intial screen appear
for touched had Suessfull to get touch and calbration at firmware flashing
stage complete.  But when I run upload  code your  code after seting
brightness function. Calibration and touch code gives continuous raw, x, y
reading ;not getting stopped like ghost touch, would you please give code
to filter those ghost touches. This will simplify my learning.

Thank you,

vednildip, 2024-12-16:
Finally  work  through  LovyanGFX   and  touch  function  in  giving  perfect  results . So  this topic  can  be  closed

closed 2024-12-16 by vednildip