Xinyuan-LilyGO/T-HMI #32: Problem with RTC

mas3058, 2025-01-31:
I have been scratching my head.  I wanted to add the time and date to my sketch.  I wrote the Time and Date to the RTC no problem.  Read and updated my sketch all worked well.  Until I removed the USBC cable and plugged it back in.  Then I was seeing the 1970 as the year. I have a LiF battery connected to VBat. its approximately 4.2vdc unloaded.  Below is my sketch. I tried the same on the T-Display S3 touch with an external battery it works fine.  I have a few T-HMI on order.  Am I missing something.

#include <ESP32Time.h>
ESP32Time rtc;
void setup() {
  Serial.begin(115200);
delay(5000);
  // Check if the RTC has been set before
  if (rtc.getYear() == 1970) { // Default year if RTC is not set
          // Set the time and date (for example purposes)
          printCurrentDateTime();
          rtc.setTime(0, 0, 12, 1, 1, 2025); // Second, Minute, Hour, Day, Month, Year
          Serial.println("RTC time set.");  
  } else {
          Serial.println("RTC time already set.");
          printCurrentDateTime();
  }
}

void loop() {}

void printCurrentDateTime() {
  Serial.print("Year: ");
  Serial.println(rtc.getYear());
}

mas3058, 2025-01-31:
No idea what markdown is.

mas3058, 2025-01-31:
first time posting anything on git

mas3058, 2025-01-31:
Looking at the schematic of the T-HMI board I can see where the battery plugs in but I don't see what pin it goes to on the ESP32S3. Can anyone help?

lewisxhe, 2025-02-06:
ESPTime is used to set the time library. If the power is off, all the set time will be lost. You must keep the power on. When you unplug the USB, the chip will switch from USB to battery power. The power will be off for a moment. Maybe you should synchronize the time from the network and then turn off the network.

mas3058, 2025-02-09:
Maybe you misunderstood me.  I thought I was clear.  I will resolve this when I get a new batch of T-HMI modules in, that I ordered.

I don't have a problem with the RTC.  I can store the date manually and recall the date and time in my sketch.  This all works fine.  I have an external battery plugged into vbat. I thought that the vbat connector was supposed to power the RTC even if the main power is turned off.  I thought the RTC would retain the date and time.  So the next time I powered up the T-HMI module the display would show the correct date and time.

I am now using an external RTC DS-3231 with its own battery.  This works fine but unnecessary if the ESP32S3 has its own built in RTC.  When I receive the new T-HMI modules I will test this again. maybe something on my T-HMI is damaged.   I will let you know.

mas3058, 2025-03-17:
Looking closely at the schematic I realize the +5 & VBAT will never be able to retain the RTC time when the main power is turned off and they are not designed to do that.  
VBAT is used to connect an external Battery for running the T-HMI module
+5 connector is used if +5vdc is needed for your project.  

The RTC section in the ESP32 is powered by pin20 vdd3p3 and this signal is tied directly to the main power.

It would have been nice if pin 20 went to a connector that you could attach a CR2032 coin battery that would keep the RTC circuit powered up. Allowing the RTC to keep the time going when main power is off.  

for now I am just using an external I2C RTC module DS3231. It works fine but not like I wanted.

mas3058, 2026-02-02:
Hi Dakkaron,

Thanks for chiming in.  No my specific problem has to do with the T -HMI module.  It is wired incorrect.  There is no internal battery backup for the esp-32S3 module.  As soon as you unplug the usbc power it kills power to the RTC.