Xinyuan-LilyGO/T-HMI #34: How to Power on T-HMI using external switch

VintageSlots2, 2025-05-08:
I have a T-HMI board that is powered by battery.

I am designing a custom case and I want to have a momentary push button to turn the device on / off.  I want it to act just like if someone is pushing the on / off button that is on the board.

I was hoping to just solder the switch to the DELLP_OFF (PZ) connectors that are on the board. 

That does NOT work.

If I unsolder the reed switch (no desire / need to have a magnet turn on the device) and connect it to those points it does work.

Since I will be modifying a lot of boards I would rather not have to do this.

What am I missing here? 

If I short the PX and PY they act exactly like the physical RST and BOT buttons.

Sure seems like the ON / OFF button should work the same way.

Thanks

[image]
[image]

lewisxhe, 2025-05-09:
From the schematic diagram, we can see that there are three activation methods.

You must use a battery or external 5V input for testing. It will not work if you use USB-C power supply.
[image]

VintageSlots2, 2025-05-10:
OK, so where do I connect the external switch leads  in order to turn the device on and off? So I don't understand your comment.

lewisxhe, 2025-05-10:
Here is a simple example that tells you how to turn the device on and off, all of which are software controlled
https://github.com/Xinyuan-LilyGO/T-HMI/tree/master/examples/power_on

VintageSlots2, 2025-05-11:
I know how to turn it off from software. 

What I want to do is turn it on / off from an EXTERNAL button.

I am putting the unit is a case and I don't want to use the buttons on the board because I want the case to be sealed from the weather as much as possible. I know it will not be IP65 rated.

I thought I should be able to hook up a push button (normally open) switch to the PZ connector on the board. 

That does NOT work. Seems like a design flaw.

I can get it to work by soldering the wires to the reed switch. I suppose I could also solder the wire to the existing on/off button which I don't want to be.

Kind of ugly but it appears to work.

Here are some photos.

[image]
[image]
[image]

lewisxhe, 2025-05-12:
Well, the external switch needs to be soldered at these two points to achieve the function of turning the machine on and off.
[image]